ThingPulse/espaper-client #2: Not yet compatible with Arduino SDK 2.4

realholgi, 2018-01-10:
It compiles with Arduino SDK 2.4, but no access point is spawned.

With SDK 2.3 I can configure the board.

squix78, 2018-01-11:
Thank you so much for pointing that out. I wasn't aware that 2.4.0 has been released. I will look into that ASAP

squix78, 2018-01-11:
I installed the Arduino/ESP8266 2.4.0, pressed the user button upon startup and the access point started. I could connect and configure without problems. Could you please try again with V2.4.0. Can you please post the serial log output here in case the problem still occurs?

realholgi, 2018-01-11:
Ok, how do I use use serial log on this board? Until now I only got garbage output...

[image]

squix78, 2018-01-11:
You have to set the correct baud rate. In this case 115200 on the serial
console