ThingPulse/espaper-client #3: HTTP CODE 500 on configured device

realholgi, 2018-01-10:
I have a Test device configured with id 34f7021d-73ae-47f4-bfd2-93b8506bc548 and the corresponding key.

[image]

On the Display I can only read "Device not found or access not allowed. Please configure this device on espaper.com HTTP CODE 500."

Interestingly the Client version is obvoiusly submitted. Hmm.

Any Ideas?

squix78, 2018-01-11:
I looked at your application "Wu" configuration. You entered a "strange" value in your wunderground configuration. The idea is to use the autocomplete look up to enter a location (at least that was my idea;-)). May I ask where you have that value from? Is it a private station? However, I will add error handling to catch such problems during rendering...

realholgi, 2018-01-11:
Ok, obviously I missunderstood the field meanings and that there is an autocomplete was not clear to me. In the end I tried to use private station the pws:IKARLSRU358 . I this somehow possible?

squix78, 2018-01-11:
In case of configuration problems the display will show an error message indicating that there might have been a problem with the configuration. In addition I opened a new issue to allow entering of stations in the format pws:IKARLSRU358: Please track this here: #10 
I will close this issue, since we now have a better error handling. The pws format will be tracked separately

closed 2018-01-11 by squix78