ThingPulse/espaper-client #11: Delete on Server fails

realholgi, 2018-01-11:
If I press the delete Button on a device, I see a message "Internal Server Error" and the device is not deleted.

[image]

realholgi, 2018-01-11:
same for applications

squix78, 2018-01-11:
Deleting devices should now work. Cause for the problem was referential integrity: with each request from the device a log entry is created which has a reference to devices. Now deleting the device will cascade also to the log. Introducing cascading delete for applications would also delete the device and I don't think this would be expected behavior. We probably should show a message saying that there is a device using this application and thus the application cannot be deleted. What do you think? And can you confirm that deleting the device now works?

realholgi, 2018-01-11:
No sorry, still "Internal Server Error" on deleting Device 12 and Application 18. Order of deletion does not matter.

realholgi, 2018-01-11:
I can delete an application, it it is not used in any device.

But now I have a device left, I cannot delete even with no application selected...

squix78, 2018-01-11:
Can you please try again? The new database constraint was not applied properly before.

realholgi, 2018-01-11:
Yay! works now!

closed 2018-01-11 by realholgi