ThingPulse/espaper-client #7: idea: barcode for guest wlan

jwende, 2018-01-11:
generate a nbarcode on the server that can be scanned and used on wifi enabled devices (ipad, phones) to connect to your guest wlan

squix78, 2018-01-11:
Hi Joerg

Interesting idea! Can you point me to a specification or a name of this feature? I wasn't aware that you could scan QR(?) codes to set the WiFi credentials. One implication though: your (guest) WLAN credentials would be stored in the cloud...

jwende, 2018-01-11:
German web articles: 
http://praxistipps.chip.de/wlan-zugang-per-qr-code-teilen_20247
https://www.smart-wohnen.de/haus-garten/artikel/per-qr-code-wifi-schnell-und-sicher-freigeben/
https://www.heise.de/ct/ausgabe/2016-10-Raspberry-Pi-als-Gaeste-WLAN-Automat-3185167.html

closed 2019-06-09 by stale[bot]