ThingPulse/espaper-client #8: idea: room booked message

jwende, 2018-01-11:
similar to the article in the last c't (2/2018) magazine
https://www.heise.de/ct/ausgabe/2018-2-Drahtloses-Tuerschild-mit-E-Paper-Display-3929847.html

squix78, 2018-01-11:
I read that article as well! This feature is already implemented, but correct me if I'm wrong. Just pick ICal calendar and choose "Meeting Room". Don't forget to add an ICAL url, e.g. to your google calendar. Or did I misunderstand your idea?

jwende, 2018-01-11:
ok - I missed the details of the ical setup. Need to find a 'public' calendar (does it work with google ?)
But one extension (I've seen this at one of my customer locations): use one of a button to create a confirmation message - or the room will change its state from "booked" to "available"

squix78, 2018-01-13:
Yes, the calendar app works with Google. Just go into the settings of a calendar. There you can find a iCal link. Copy and paste it into the Calendar App settings...
About the confirmation message: cool idea. With the current hardware we have a limitation, that we can't read out button states when the device is sleeping. This is a limitation of the ESP8266 but could be solved with external hardware or a ESP32

closed 2019-06-09 by stale[bot]