ThingPulse/espaper-client #25: Test speed (i.e. battery life) improvements with persistent BearSSL sessions

marcelstoer, 2019-01-01:
https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/bearssl-client-secure-class.html?highlight=rtc#tls-sessions

With ESP8266 Arduino-core 2.5.0 we may be able to serialize the BearSSL::Session object to RTC RAM and restore it before trying to (re)connect. Should give a significant speed (and thus battery life) boost.

squix78, 2019-01-02:
In this context I would love if we could extend the headers the client sends by a field which measures the full awake time of the last cycle.

marcelstoer, 2019-01-02:
-> #26

marcelstoer, 2019-07-15:
We should keep this open and also consider the other recommendations in that great documentation page.

> it is recommended that all TLS/SSL sketches to run at 160 Mhz and not the default 80 Mhz.

wrt energy: higher CPU speed vs. shorter air time

closed 2021-06-02 by stale[bot]