ThingPulse/espaper-client #32: Weather Widget

schroepfer, 2019-03-10:
Hello,
i want to user the weather widget.
I filled the API-Key and City Name, butt all my known City-IDs doesn't work!?
I get only Zürich.  ;-)
Regards
Jürgen

squix78, 2019-03-10:
Hi Jürgen. 
So just to be sure, you have a valid OpenWeatherMap key and tried to enter a city name but the lookup didn't give you a result? Can you give me an example city name you tried?

Thank you,
Dani

schroepfer, 2019-03-10:
Yes, in other applications the key works. I tried "Melsungen".
Regards
Jürgen

squix78, 2019-03-10:
Hi Jürgen

Thank you so much for pointing this out. Can you please check again? I
think I fixed it now...

Kind regards,
Daniel

schroepfer, 2019-03-10:
No, it doesn't work:

[image: Screen Editor 2019-03-10 19-15-48.jpg]
All fields are green, City ID is red.
When i use the widget i get Zürich.

squix78, 2019-03-10:
There is something fishy indeed. The first time I open the dialog I get:
[image]
Then afterwards:
[image]

I can't see the image you attached. Does it look the same?

schroepfer, 2019-03-10:
Yes, exactly.

closed 2019-10-06 by stale[bot]