ThingPulse/espaper-client #36: Can't Set Weather Widget City Name

wabmester, 2019-04-21:
I'm able to use the weather widget, and it correctly fetches the weather data, but two related issues:

1) My city name is long ("Bainbridge Island") and it wraps around, over the temperature.

2) I can't edit the city name in the app settings.   It keeps getting overwritten by the json name associated with the city id (5816320).    This may be similar to another open issue where the city name field was misbehaving.

datacute, 2019-07-28:
I've just started using this today, and discovered that you can 'ungroup' the elements, then double click on the city name element, and change it to be constant text instead of read from the data.

wabmester, 2019-09-16:
Nice work-around.   When I tried it, I needed to delete the city name element and replace it with static text, otherwise it still reverted to the fetched name.

marcelstoer, 2019-09-16:
Thanks for raising this issue. We're not sure what the best option is here but it _is_ probably not intuitive that the city name you configured is not the one displayed. So far, the sole purpose of that configured name is to look up the location id that OWM needs (their API also supports query by name or lat/long but results are then ambiguous).

Do you think we should change that?

The procedure to edit it in your screen is as follows.

**Ungroup the widget**

**Double-click the city name**
[image]
**Replace the ${} expression with static content**
[image]
**Save**
[image]

closed 2020-03-28 by stale[bot]